m-samland/spherical #61: [Feature]: example or test of end-to-end IFS reduction pipeline

lwelzel, 2025-05-14:
### ✨ Description of the Feature

I am running into an issue with the spectral cube extraction and calibration steps in the pipeline and am unsure at what step the problem occurs. To verify the correct execution of the reduction pipeline, the step-by-step output of an example reduction would be helpful. This could e.g. be the step-by-step output from running examples/ifs_reduction_template.py on a specific observation.

### 🔖 Proposed Solution

Since the tables/database are already accessible via the zenodo archive, verification could either be against example reduction files also stored on the zenodo archive or against summary statistics from the output. Ideally this would be an automated test in /tests.

This might be easier if the spherical.pipeline.ifs_reduction.execute_target() function is broken up into its constituent steps.

m-samland, 2025-05-17:
I started with some minor refactoring and moving things away from the giant ifs_reduction.py module. I want everything that is in an `if`-statement at the moment to be its own step in its own file (similar to the JWST pipeline), and have consistent logging output throughout. Every step should also check if the expected files at the end were producted.

I haven't had the time to start on this yet, since with version 2.0 I wanted to enable IRDIS and polarimetry in the database framework, and to at least be able to download the data for those observations, plus get rid of all known bugs in the tables. That should be done now, so it would be nice in the next release v2.1.0 to focus on refactoring the pipeline into something clean and modular. That way, it would also be possible to implement steps that are shared between IRDIS and IFS to use the same routines cleanly (like for example the star center finding and photometric calibration). 

Do you think you would have some time to refactor out any of the steps or do you want to wait until I get around to do it? I'll be pretty busy in the near-term though. If you let me know what data you're struggling with I can also have a look at the particular failure case. I have reduced quite a lot of data with the newest version and have not yet run into issues, but I also just fixed some edge-cases again.

lwelzel, 2025-05-19:
That is great to hear, I can have a look at separating the steps in the reduction pipeline, however, I want to make sure that everything works as expected on my side before I start with that.

m-samland, 2025-05-20:
That's great! Note that with v2 you have to download the new tables that I have uploaded to Zenodo, because some columns have changed. They are compatible with the new example templates.

m-samland, 2025-05-20:
> ### ✨ Description of the Feature
> I am running into an issue with the spectral cube extraction and calibration steps in the pipeline and am unsure at what step the problem occurs. To verify the [...]

The main problem with testing the IFS pipeline wavelength calibration and cube building is that it takes significant computational resources to run, so it is hard to do this with CI on Github. Ideally, these tests should be in the CHARIS repository, since we should assume them to be working in spherical. The problem is that both steps take significant resources and the reduced calibration files are very large too. That kind requires running the tests locally offline (but we can have a set of standard input files that we test offline, potentially on a Zenodo archive).

I think all the later steps should be much easier to test, either with dummy data we make, mock patches, or again some sample outputs, e.g. containing one extracted flux, center and coro frame.

lwelzel, 2025-06-15:
Related to this, it might be a good first step to agree on a common 'testing' observation (or a set of them for different cases) so that we can run tests consistently (even if only locally).

Starting out, do you have a suggesting for a 'well behaved' observation? Good-to-haves for such an observation would be:
- has flux, center, and coro
- good observing conditions
- is relatively small (small total file-size)
- well understood system (e.g. HR 8799, beta Pic)

This would then also be a good point to add a full pipeline integration test with that observation which saves out to '/spherical/tests/data/... '. We can disable this test/mark it as skip by default or disable it for GitHub CI.