m-samland/spherical #62: [Feature]: Add dedicated logging to the IFS pipeline

lwelzel, 2025-05-14:
### ✨ Description of the Feature

Both spherical and the external libraries (CHARIS pipeline, SPHERE tools) log to the console by default. Understanding, tracking progress of, and identifying problems with a pipeline run would be easier if logs were piped to the output folders associated with the pipeline steps. This would then allow also only printing errors/warnings (or progress) to the console. 

### 🔖 Proposed Solution

We can use the default python logger or a dedicated logging package to directly replace print statements currently in the code and intercept messages from the external pipelines and redirect them to a dedicated log file. 

Logging could either be persistent (e.g. per processed observation, preferred by me) or we can keep one log file of the last pipeline execution which is overridden if the pipeline is re-executed.

m-samland, 2025-05-20:
I would also prefer one log file per processed observation. There can also be warnings that do not crash the observation directly, but may be useful for scientific interpretation of the results.

m-samland, 2025-06-25:
Huge update included on the feature/logging branch. I'm sure there are things not perfect with it yet, but it should mostly cover the logging requirements. Most of the functionality is in the new pipeline.logging_utils module, and I made a template for how the logging refactoring was done: https://github.com/m-samland/spherical/blob/feature/logging/.github/LOGGING_GUIDELINES.md.