m-samland/spherical #90: [Bug]: Finding data of binary star

tomasstolker, 2025-07-28:
When searching for IRDIS data of HD131399A it does not find any, whereas there are multiple datasets in the ESO archive. When searching for data of HD131399 it does find the datasets of the primary star. Not sure if this is a specific problem for this visual binary system or happens in other cases as well.

Perhaps also causes by a double entry in SIMBAD:
http://simbad.u-strasbg.fr/simbad/sim-basic?Ident=HD+131399
http://simbad.u-strasbg.fr/simbad/sim-basic?Ident=HD+131399A

tomasstolker, 2025-08-06:
Are these changes also part of the `develop` branch? I am up to date with that branch (1.1.2.dev19+gb457863)

m-samland, 2025-08-06:
Oh yes they are, if you're using "pip install -e .", it doesn't update the version number if you pull a new version. Only if you re-run the installation command, so if you're on the develop it should be 2.1.1. Could you confirm again?

Just to make sure I understand the context, you are using the `database.target_list_to_observation_table(target_list)` syntax from the example script when you encounter this issue?

If this is a persistent issue, I have to dig a bit. It might make sense to simply strip "A" from the end all target names fed into the function. I think SIMBAD will always treat the name of the star itself as the primary component of a binary?

commit 77d8073f (2025-08-06): Fix #90
closed 2025-08-10 by m-samland